robotology/whole-body-controllers #8: Shall we use Travis for dealing with different Simulink versions?

gabrielenava, 2018-01-09:
It has always been a problem to deal with different Simulink versions because a model created with a newer version cannot be opened using an older version. Same happens if a model created with an old version is opened with a newer Simulink. 

In [WBIToolbox-Controllers](https://github.com/robotology-playground/WBI-Toolbox-controllers) the proposed solution was to [duplicate each model](https://github.com/robotology-playground/WBI-Toolbox-controllers/tree/master/controllers/torqueBalancing) so that it is possible to open it with different Simulink versions. I don't like this solution for two reasons:

- It is difficult to mantain all models even if the exporting procedure is straightforward. Furthermore, users are often confused because they do not know which model to run;
- If I'm a user of an old Simulink version, there is no way for me to know if the model I'm using is the most updated one. 

In this repo, I came up with a [simpler solution](https://github.com/robotology-playground/wholeBodyControllers/blob/master/config/export_WBC.m): all models are written in `2016b` which is the Matlab version that is used inside the lab. Plus, a [legacy](https://github.com/robotology-playground/wholeBodyControllers/tree/master/legacy) version of the models is saved in the lowest supported version, i.e. Matlab 2014a.

### Why not save directly the models in version 2014a?

Because this implies the user must run the [export script](https://github.com/robotology-playground/wholeBodyControllers/blob/master/config/export_WBC.m) every time he need to push a modification in the repo, and I do not trust the user (e.g., I already pushed one commit using Matlab 2017b because I forgot to export once).

### So, what we can do?

A solution can be: the user can push any version he wants. Then, Travis can automatically export all models into version 2014a. But I do not really know if this is feasible.

cc @traversaro @diegoferigo

traversaro, 2018-01-09:
I don't think there is a way for Travis to run Simulink. On the other hand, we could implement the proposed behaviour using our internal Jenkins infrastructure.

diegoferigo, 2018-01-10:
@traversaro I know one way, but it might not be compliant on Matlab's licence and I was gently asked not to mess with it :sweat_smile:

As discussed with @DanielePucci back in December, I'm going to develop a bash script which will be handled by our internal Jenkins infrastructure.

The implemented rationale will be the following:

Let's take the minimum version of Matlab / Simulink we want to support [*]. Every commit triggers Jenkins which will:

- Clone the repository
- Open with its Matlab version (which will be the most recent we have, 2017b) the controller
- Override the committed version with the one exported matching the minimum version supported
- Commit the changes

[*] We still need to figure this out. Me and @gabrielenava discussed briefly about issues with scopes during the exporting. In my opinion we should select the latest Matlab version which doesn't mess around the model and is less recent than the oldest version we have in the lab.

gabrielenava, 2019-02-28:
For the moment I'll close this issue, as for dealing with different Simulink versions until now it suffices to use the [export_WBC](https://github.com/robotology/whole-body-controllers/blob/master/config/export_WBC.m) script.

closed 2019-02-28 by gabrielenava