robotology/whole-body-controllers #13: Boost data processing for data coming from the Simulink model

gabrielenava, 2018-01-23:
With [this commit](https://github.com/robotology-playground/wholeBodyControllers/commit/b82317a0e9eeccdb7bdb3ba22e81449e2b4d43b6), I added a callback in every model to a function like [this one](https://github.com/robotology-playground/wholeBodyControllers/blob/WBT3/controllers/torqueBalancingYoga/stopTorqueBalancingYoga.m). The function is called every time the `terminate` button is pressed. The purpose was to remove local paths that are added for running a specific simulation, e.g. Yoga or Standup. The reason is that the local path may contain a functions with the same names and this may be dangerous (one doesn't know which function is taken). 

However, I noticed that this callback function is run only after the `to workspace` block updates the workspace. I realized that we can then use this function to plot data, save the workspace, and so on, and this may be super useful for processing data while on the robot, compare two different simulations and so on.

gabrielenava, 2018-01-29:
I will try to implement this during this week. The Idea is the following:

- run a demo;
- terminate and collect data, plot data;
- decide whether to save or discard data.

@S-Dafarra may be useful also for you for collecting data on the robot.

gabrielenava, 2018-02-02:
I think I will implement this soon for collecting data for IROS.

diegoferigo, 2018-02-04:
@gabrielenava In my opinion we should exploit this features and dump all the data we can gather from the robot using simulink to the workspace in a structured way. With this I mean that the initialization script could contain a flag to enable / disable data logging, and all the blocks that export data (scopes mainly) are configured in such a way to fill a class' object.

This class can be developed separately and might provide useful methods for automatically post-processing and plotting data. If needed, serializers to different file formats (csv, ...), e.g. for committing datasets, might be integrated easily.

We can have a f2f meeting for discussing this more in detail (remember: you applied _priority low_!)

gabrielenava, 2018-03-21:
closing as a duplicate of https://github.com/robotology-playground/whole-body-controllers/issues/20

closed 2018-03-21 by gabrielenava