robotology/whole-body-controllers #16: Whole Body Control

lpenco, 2018-02-06:
I'd like to control all the joints of the iCub included the wrist and the head. 
I changed the parameters in gains.m but I get this error. What else do I have to change?

Error evaluating 'InitFcn' callback of block_diagram 'torqueBalancingR2015bXsens'.
Caused by: Dimensions of matrices being concatenated are not consistent.

gabrielenava, 2018-02-06:
mmh...ok, the error is related to a dimension mismatch that I'm afraid is caused by the gains and references inside [initStateMachine.m](https://github.com/robotology-playground/WBI-Toolbox-controllers/blob/master/controllers/torqueBalancing/app/robots/icubGazeboSim/initStateMachine.m) and [initRefGen.m](https://github.com/robotology-playground/WBI-Toolbox-controllers/blob/master/controllers/torqueBalancing/app/robots/icubGazeboSim/initRefGen.m). Unfortunately, Simulink requires all these parameters to be loaded when compiling the model (even if not used because e.g. they are inside the `yoga state machine` subsystem), so you need to modify almost all of them. Furthermore, I'm not sure if it is possible to control joints such as the neck and the wrist using torque control. This may be tricky, I'll double check it tomorrow.

lpenco, 2018-02-07:
yeah it works by changing all the yoga parameters. Thank you! :)

closed 2018-02-07 by lpenco