robotology/whole-body-controllers #20: Automatic data processing and plotting of whole-body datasets

diegoferigo, 2018-03-03:
Following https://github.com/robotology-playground/whole-body-controllers/issues/13, it would be really nice having a matlab class or script that automatically processes data logged from Simulink (initial idea [here](https://github.com/robotology-playground/whole-body-controllers/issues/13#issuecomment-362902602)). This would require introducing a generic structure for storing whole-body data.

In this way, if the visualization section of all controllers is the same (better if shared), creating plots for publications and documentation would become trivial. We know well how much time we spend on this.

cc @gabrielenava @S-Dafarra

gabrielenava, 2018-09-05:
In view of https://github.com/robotology/wb-toolbox/issues/160, it seems that having a way to process the data after a simulation without the need of opening the Simulink gui may become important soon. 

@diegoferigo I think we should discuss if the `auto-generated` models should become very soon our standard way to run demos on the robot, and therefore address also this issue.

diegoferigo, 2018-09-05:
We never cared too much about real-time performance of our models because most times everything worked fine but probably they suffer of similar issues.

At the current stage the only limitation of the autogenerated code is gain tuning, it would require work about we store parameters in the autogenerated class and it will take some time to implement something intuitive enough.

However, we should start testing the C++ code of current model with hardcoded gains coming from the Matlab workspace from the external pc, and then move to the robot head.

gabrielenava, 2018-12-14:
@Yeshasvitvs suggested a possible workflow in order to be able to see run time plots with Simulink. The idea is to stream the Simulink data in a `yarp port`, and then use the `yarp scope` in order to visualize the streamed data. Seems an interesting idea, but the feasibility is still to be verified. On the other hand, @FabioBergonti can provide a matlab script to perform post-processing on the data coming from Simulink. The addition of this feature to the WBC may be very useful and can be discussed in the next software upgrade, too.

Yeshasvitvs, 2018-12-14:
> @Yeshasvitvs suggested a possible workflow in order to be able to see run time plots with Simulink. The idea is to stream the Simulink data in a `yarp port`, and then use the `yarp scope` in order [...]

For reference  check the [yarpscope xml](https://github.com/robotology-playground/forcetorque-yarp-devices/blob/master/ftShoe/conf/ftshoes_YarpScope.xml.txt) we have now for visualizing FTShoe data. I encoded the color scheme similar to matlab for the plots with in a scope. 

[image]

gabrielenava, 2019-03-07:
For data naming convention in `whole-body-controllers`, refer to this issue: https://github.com/robotology/whole-body-controllers/issues/64

gabrielenava, 2019-10-15:
with [this commit](https://github.com/robotology/whole-body-controllers/pull/77/commits/1b80900dc6736d40d3d4cd56687193108f651554) I've added a Matlab script that automatically locates and processes the data coming from Simulink.

I will therefore close this issue, and I will open a new issue specifically for having **runtime plots** with yarpscope, as detailed in [this comment](https://github.com/robotology/whole-body-controllers/issues/20#issuecomment-447345904), that anyways will be implemented in a future release.

closed 2019-10-15 by gabrielenava