robotology/whole-body-controllers #25: Compilation broken on master branch

gabrielenava, 2018-05-24:
I tried to compile the repo using the [robotology-superbuild](). I reverted `yarp` and `icub-main` to `master`  branch. I got the following error:

```cmake
Not searching for unused variables given on the command line.
loading initial cache file /home/gnava/Software/github/robotology/robotology-superbuild/build/robotology/whole-body-controllers/CMakeFiles/YCMTmp/whole-body-controllers-cache.cmake
CMake Error at cmake/RegisterMdl.cmake:60 (message):
  Folder torqueBalancingYoga_grt_rtw not found.
Call Stack (most recent call first):
  torque-controllers/momentum-based-yoga/CMakeLists.txt:1 (register_mdl)

-- Configuring incomplete, errors occurred!
See also "/home/gnava/Software/github/robotology/robotology-superbuild/build/robotology/whole-body-controllers/CMakeFiles/CMakeOutput.log".
CMakeFiles/whole-body-controllers.dir/build.make:103: recipe for target 'robotology/whole-body-controllers/CMakeFiles/YCMStamp/whole-body-controllers-configure' failed
make[2]: *** [robotology/whole-body-controllers/CMakeFiles/YCMStamp/whole-body-controllers-configure] Error 1
CMakeFiles/Makefile2:1103: recipe for target 'CMakeFiles/whole-body-controllers.dir/all' failed
make[1]: *** [CMakeFiles/whole-body-controllers.dir/all] Error 2
Makefile:94: recipe for target 'all' failed
make: *** [all] Error 2

```

cc @diegoferigo

gabrielenava, 2018-05-24:
I suspect I should generate the C++ code from Simulink in order to compile. I will try to create a "fake" autogenerated folder to see if in this way the repo can compile, but we should find a way to fix this permanently.

traversaro, 2018-05-24:
Mhhh for the time being I think you can just comment out most of the CMakeLists.txt

diegoferigo, 2018-05-24:
@gabrielenava Right, I didn't think the superbuild was executing the dummy cmake. Try to add after `project()`:

```cmake
option(WBC_EXPORT_AUTOGENERATED "Enable the target to export code generated with Simulink Coder" OFF)

if(NOT WBC_EXPORT_AUTOGENERATED)
    return()
endif()
```

gabrielenava, 2018-05-28:
I follwed the suggestion [here](https://github.com/robotology/whole-body-controllers/issues/25#issuecomment-391790148) but anyway I was not able to compile the repo. I got this error:

```cmake

make[3]: *** No rule to make target 'install'.  Stop.
CMakeFiles/whole-body-controllers.dir/build.make:72: recipe for target 'robotology/whole-body-controllers/CMakeFiles/YCMStamp/whole-body-controllers-install' failed
make[2]: *** [robotology/whole-body-controllers/CMakeFiles/YCMStamp/whole-body-controllers-install] Error 2
CMakeFiles/Makefile2:100: recipe for target 'CMakeFiles/whole-body-controllers.dir/all' failed
make[1]: *** [CMakeFiles/whole-body-controllers.dir/all] Error 2
Makefile:94: recipe for target 'all' failed
make: *** [all] Error 2

```

traversaro, 2018-05-28:
You need to put the 
~~~
# Fake installation. 
install(CODE "MESSAGE(\"This repository simply acts as a container\")")
~~~ 
before the `return()` .

gabrielenava, 2018-05-28:
This was also my conclusion. I tested with:

```cmake

if(NOT WBC_EXPORT_AUTOGENERATED)
    # Fake installation. 
    install(CODE "MESSAGE(\"This repository simply acts as a container\")")
    return()
endif()

```

It is working. Closing the issue.

gabrielenava, 2018-05-28:
The fix is at https://github.com/robotology/whole-body-controllers/commit/234bb90682d7dfedf997ec37bf0f73425ba2566a

closed 2018-05-28 by gabrielenava