robotology/whole-body-controllers #29: torqueBalancingYoga and YARP 3

diegoferigo, 2018-06-14:
I am experiencing a strange behavior with the following combination of software:

| Project | Version |
| --- | --- |
| yarp | 3.0.0 |
| wb-toolbox | latest devel + yarp3 fixes (https://github.com/diegoferigo/WB-Toolbox/commit/4ee6e10fc5ab35c06ae96392b64ea654dae904a1) |
| icub-main | latest devel (https://github.com/robotology/icub-main/commit/4f5881acafab693f763dd277f87a8996fdea01a3) |
| idyntree | latest master (https://github.com/robotology/idyntree/commit/94cee9db71266d65445a7a752342fa473159faaf) |
| gazebo-yarp-plugins | latest devel (https://github.com/robotology/gazebo-yarp-plugins/commit/31707215ad25368ede2c799b253b89821cb50600) |
| codyco-modules | latest devel (https://github.com/robotology/codyco-modules/commit/e1965bb4367ea3af07b53122ab26525c94847d56) |

I am pretty sure that with the latest devel of wb-toolbox, that compiles against yarp devel pre-3 release, the controller was working fine. I remember I tested the generated yoga controller.

The new behavior is the following:

[image]

I doubt this is a problem of the controller itself. More probable the issue is somewhere else, hard to tell at the moment.

cc @gabrielenava @traversaro @DanielePucci

diegoferigo, 2018-06-14:
Since this period's updates broke many things I tried to start with a new and clean isolated environment in order to slowly remove variables from the big picture. After some digging I realized that the uncommented robot name in the `initTorqueBalancingYoga.m` was `icubGazeboSim` and not `iCubGazeboV2_5` -.-

Everything works fine now.

closed 2018-06-14 by diegoferigo

traversaro, 2018-06-14:
@gabrielenava Do you think it make sense to not overwrite the `YARP_ROBOT_NAME` of the system in the file that we commit to the repo? Or there is some problem in Matlab to access the `YARP_ROBOT_NAME` set in the enviroment?

gabrielenava, 2018-06-14:
> @gabrielenava Do you think it make sense to not overwrite the YARP_ROBOT_NAME of the system in the file that we commit to the repo? Or there is some problem in Matlab to access the YARP_ROBOT_NAME set in the environment?

I guess there are no issues. It suffices to use the matlab command `getenv`. I  can open an issue for this.