robotology/whole-body-controllers #31: Improve documentation on How-to-run-torqueBalancing-on-real-iCub

lrapetti, 2018-06-15:
Some passages of the [How-to-run-torqueBalancing-on-real-iCub](https://github.com/robotology/whole-body-controllers/blob/master/doc/How-to-run-torqueBalancing-on-real-iCub.md) are missing or not clear for new users.

* The SIMULATION reference in the title is tricky (it is the same as [How-to-run-torqueBalancing-simulations-with-iCub](https://github.com/robotology/whole-body-controllers/blob/master/doc/How-to-run-torqueBalancing-simulations-with-iCub.md) to work in simulation with Gazebo)

* The YARP_ROBOT_NAME is set in the `init` function (e.g. [InitTorqueBalancingYoga](https://github.com/robotology/whole-body-controllers/blob/master/torque-controllers/momentum-based-yoga/initTorqueBalancingYoga.m#L32)), so it has to be changed accordingly to the robot one wants to use.

* It is not specified which passages have to be executed with the robot on-flight or on-the-ground. (e.g. custom position -> on-flight, run the module -> on-gorund)

@gabrielenava @traversaro

gabrielenava, 2018-06-19:
I tried to address this issue with these commits:

https://github.com/robotology/whole-body-controllers/commit/da71167ceb03be70970f90fd4446a2309d7986dc

https://github.com/robotology/whole-body-controllers/commit/4a17108437a9fdbde551add308bf16ad4c02851a

being a very simple modification I didn't open a pull request. @lrapetti let me know if further modifications are required.

lrapetti, 2018-06-19:
Good! Now everything is clear to me!
I am closing the issue.

closed 2018-06-19 by lrapetti

traversaro, 2018-06-19:
I think we should also specify a "suitable home position" for the balancing, in 
> Bring the robot in a suitable home position (e.g. `$ yarpmotorgui --from homePoseBalancing.ini` and then select a custom position by clicking on `Global Joints Commands/Custom postions`.