robotology/whole-body-controllers #38: Check if the Simulink required integration step is respected

gabrielenava, 2018-09-05:
In view of https://github.com/robotology/wb-toolbox/issues/160, it seems that there might be an issue with the Simulink gui that impairs the achievement of the desired integration step.

As first thing, we need to verify if there are differences between running the standard demos (Yoga, standup) from command line or from the Simulink gui. 

@FabioBergonti if you already implemented a piece of code that allow to verify if the Simulink integration time is respected, can you please add it to this issue? I will then proceed to integrate it inside all the models.

FabioBergonti, 2018-09-05:
Here what I did
I've added this block in my simulink model
[image]
Then I evaluate `YARP_time0`
``` matlab
YARP_time0 = YARP_time.signals.values - YARP_time.signals.values(1);
```
After that some plots
#### Yarp Time VS simulink time
``` matlab
figure
hold on
plot(YARP_time0,0:length(YARP_time0)-1,'o')
plot(tout,0:length(tout)-1,'o-')
xlabel('Time [s]')
ylabel('Iteration')
grid on
legend('Yarp Time','Simulink Time')
```
[image]

#### Delta Yarp Time
``` matlab
figure
plot(tout(1:end-1),diff(YARP_time0),'.-')
ylabel('Diff(Yarp Time) [s]')
xlabel('Absolute time [s]')
```
[image]

#### Distance between huge Delta Yarp Time
``` matlab
figure
Acc=diff(YARP_time0);
plot(diff(YARP_time0(Acc>0.018)),'.-')
ylabel('Distance [s]')
```
[image]

gabrielenava, 2018-09-11:
Thanks @FabioBergonti! I added the checks in both the `Yoga` and `standup` models with [this commit](https://github.com/robotology/whole-body-controllers/commit/cb770c073f478f83d808a15d98b0ee1deed72c3e)

gabrielenava, 2018-09-14:
I tested the Yoga++ demo on `iCubGenova02`, both from the Simulink GUI and from command line. As you can see, I got the same results @FabioBergonti already had for the jumping demo:

**From Simulink GUI, Yarp Time Vs. Simulink Time**

[image]

Looking closer:

[image]

**From Matlab command line, Yarp Time Vs. Simulink Time**

[image]

[image]
 
I perceived the demo was also qualitatively different (faster) when launching it from the command line. Therefore I close this issue. The default method for running Simulink controller will become to launch the controllers from command line.

closed 2018-09-17 by gabrielenava

Yeshasvitvs, 2019-07-03:
@venusPasandi FYI