robotology/whole-body-controllers #39: Update all models to launch Simulink from command line

gabrielenava, 2018-09-14:
In view of https://github.com/robotology/wb-toolbox/issues/160 and https://github.com/robotology/whole-body-controllers/issues/38, it is necessary to modify all the startup procedure of the Simulink controllers. The idea is to create a Matlab function named `startTorqueBalancingxxx.m` and the open a (static) gui that can handle the Simulink model without opening it.

traversaro, 2018-09-14:
As you may know, there is no guarantee that this new way of launching the controllers will not stop working in new versions of MATLAB, depending on how the (unknown) internal details of MATLAB change. For this reason I think it would be ideal if there is a way to always inspect the discrepancy between the real time and the simulation time, eventually printing a warning/error or stopping if the maximum delay is reached. This will ensure that they will be no regression w.r.t. to this in the future.

gabrielenava, 2018-09-17:
I agree. I will also include a warning or error

gabrielenava, 2018-09-17:
I've found this example: https://www.mathworks.com/matlabcentral/answers/97557-is-it-possible-to-start-and-stop-a-simulink-simulation-from-inside-a-gui I will try to follow this procedure.

gabrielenava, 2018-09-17:
Another interesting review of 3 different methods: https://blogs.mathworks.com/pick/2012/06/01/use-matlab-guis-with-simulink-models/

gabrielenava, 2018-09-20:
A first prototype of the GUI have been created in [this branch](https://github.com/robotology/whole-body-controllers/tree/noSimulinkGui). There are still a couple of things I want to solve before merging in `master`, but I already tested the GUI on `iCubGenova02`. These are the results:

**Yoga ++, from Simulink GUI**

[image]

[image]

The problem is clearly present. 

**Yoga ++, from the new static GUI**

[image]
[image]

By running Simulink with the new static GUI, the problem is solved and the difference between two time steps is always closed to 0.01 s. A similar result is obtained by running Simulink from command line:

**Yoga ++, from command line**

[image]
[image]

So, it seems the new GUI works! Here is a picture of how it looks like:

[image]

gabrielenava, 2018-12-14:
The [PR](https://github.com/robotology/whole-body-controllers/pull/42) is ready to be tested on the robot. Closing this issue, as the troubleshooting from now on will be done inside the PR.

Concerning the GUI, at this stage it has been added to the Yoga demo only. After testing it on the real robot, we may discuss if adding it also to the `standup` and `impedance` controllers in a future release.

closed 2018-12-14 by gabrielenava