robotology/whole-body-controllers #45: Wrong torso coupling matrix

lrapetti, 2018-10-05:
The coupling matrix for the [torso](https://github.com/robotology/whole-body-controllers/blob/master/torque-controllers/momentum-based-yoga/app/robots/iCubGenova04/initStateMachineYoga.m#L447) that is used in the `whole-body-controllers` seems not compatible with the [joints order configuration](https://github.com/robotology/whole-body-controllers/blob/master/torque-controllers/momentum-based-yoga/app/robots/iCubGenova04/configRobot.m#L22).

Given the motor-joints order used in the configuration:

| index  | **joints** | **motors**|
| ------------- | ------------- | ------------- |
| 0  | `torso_pitch`  | `OB3M1` |
| 1 | `torso_roll`  | `OB3M0`  |
| 2  | `torso_yaw`  | `OB4M0` |

The transform used in `whole-body-controllers` is:
 $$ \begin{pmatrix} \text{torso_pitch} \\\ \text{torso_roll} \\\ \text{torso_yaw} \end{pmatrix} = \begin{pmatrix}  0.0 & -0.5 & 0.5 \\\ 0.0 & 0.5 & 0.5 \\\ 0.55 & 0.275 & 0.275 \end{pmatrix}  \begin{pmatrix} \text{OB3M1} \\\ \text{OB3M0} \\\ \text{OB4M0} \end{pmatrix} $$

While, accordingly to the [wiki](http://wiki.icub.org/wiki/ICub_coupled_joints) and respecting the joints convention used in the configuration of the `whole-body-controllers`, it should be:
 $$ \begin{pmatrix} \text{torso_pitch} \\\ \text{torso_roll} \\\ \text{torso_yaw} \end{pmatrix} = \begin{pmatrix}  -0.5 & 0.5 & 0.0\\\ 0.5 & 0.5 & 0.0 \\\ 0.275 & 0.275 & 0.55 \end{pmatrix}  \begin{pmatrix} \text{OB3M1} \\\ \text{OB3M0} \\\ \text{OB4M0} \end{pmatrix} $$
\
\
Probably, the error was due to the fact that the matrix was defined starting from the [wiki](http://wiki.icub.org/wiki/ICub_coupled_joints) and changing the row of the matrix without considering the motors order.

This impacts the computation of the `motor reflected inertia`

gabrielenava, 2018-10-05:
Nice! So, I recently double checked the coupling matrices and I noted the same issue related to the torso coupling matrix raws and columns (I compared the matrices with the ones in robots-configuration) and I committed the modifications (https://github.com/robotology/whole-body-controllers/commit/ac77b9905be2926defa76b8afd8f2eb73cef8b0f) but apparently I forgot to fix this. It will be done soon.

gabrielenava, 2018-10-09:
Fixed in this commit: https://github.com/robotology/whole-body-controllers/pull/46/commits/744677118d0f30f040f90ff36cdabf7306aded86

gabrielenava, 2018-10-10:
In the end, it has been fixed in this [PR](https://github.com/robotology/whole-body-controllers/commit/7ab9d86e5db9d6701ea36e2b6efe276ce0321a61). Closing.

closed 2018-10-10 by gabrielenava