robotology/whole-body-controllers #49: Shall we add the harmonic drives inertia in the motors reflected inertia

FabioBergonti, 2018-10-09:
What do you think on adding the inertia of the harmonic drives?
From the [datasheet](https://harmonicdrive.de/mage/media//catalog/category/2014_11_ED_1019659_CSD-2A.pdf) I saw their inertia and is about $0.054*10^{-4} kg m^2$ (this is the mass moment of inertia of the gearing with respect to the input)
The dimension are not so different from the motors inertia

gabrielenava, 2018-10-10:
It makes sense to me!

traversaro, 2018-10-10:
[image]

Sorry for the italian.  : ) 
We had exactly the same discussion with @DanielePucci and @fiorisi , and we agreed that it make sense to include inertia of the part of the harmonic drive that is coupled with the motor shaft in the reflected inertias, as their order of magnitude is similar.

traversaro, 2018-10-10:
cc @triccyx given that was apparently the one that sparked the discussion

FabioBergonti, 2018-10-10:
Here a cross section of an harmonic drive
[image]
1. input shaft
2. wave generator
3. flexspline
4. circular spline
5. output shaft
6. housing

We have to add **only**  the inertia of the components (1) and (2), (all the rotating components before the flexspline)

gabrielenava, 2018-10-10:
According to [this document](https://harmonicdrive.de/mage/media//catalog/category/2014_11_ED_1019659_CSD-2A.pdf) and looking at the following lines:

[image]
[image]

 it seems to me that for  the `CSD-17-2A` series the inertia value is $0.054*10^{-4} kg m^2$  that is the one that also  @FabioBergonti  suggested. I added it to the motors reflected inertia with this [PR](). Waiting for testing it on the robot.

gabrielenava, 2018-12-14:
The [PR](https://github.com/robotology/whole-body-controllers/pull/42) is ready to be tested on the robot. Closing this issue, as the troubleshooting from now on will be done inside the PR.

closed 2018-12-14 by gabrielenava

DanielePucci, 2018-12-20:
Very well done guys. Thanks a lot

traversaro, 2019-02-13:
Related Pinocchio issue: https://github.com/stack-of-tasks/pinocchio/issues/440 .