robotology/whole-body-controllers #52: [WARNING]: WBC library of matlab functions will be transformed into a package in the future release

gabrielenava, 2018-12-14:
In the current `master` branch, there is a [library](https://github.com/robotology/whole-body-controllers/tree/master/library/matlab) of matlab utility functions that are used by the controllers. However, we have other controllers in other github organizations, including [matlab-simulators](https://github.com/gabrielenava/matlab-multi-body-sim) that use local copies of these functions. I decided to convert this library in a [package](https://it.mathworks.com/help/matlab/matlab_oop/scoping-classes-with-packages.html) in order to:

-  avoid to duplicate the `library` in several repositories;
-  avoid to permanently add the `library` to the Matlab path, that may be dangerous in case there are functions with the same name.

The modification has been implemented and tested  in this [PR](https://github.com/robotology/whole-body-controllers/tree/noSimulinkGui). When merged into `master` all the standard demos `yoga`, `standup`, `impedance` should work as before. All other not officially supported demos can  work as before as well, provided that the user **adds the prefix `wbc.` to the name of all the functions of the `library` when there is a call to these functions.** E.g.:

```

function [ ] = myRandomFunction ()

   % call to one of the functions in the library
   [outputs] = wbc.nameOfFunctionInsideTheLibrary(inputs)
end

```
cc @Yeshasvitvs @aikolina

gabrielenava, 2018-12-21:
Closing. I sent an email.

closed 2018-12-21 by gabrielenava

Yeshasvitvs, 2019-06-30:
@gabrielenava The PR link is broken.

gabrielenava, 2019-07-01:
link restored, thanks!