robotology/whole-body-controllers #56: Fix broken links in main README

aikolina, 2019-01-10:
There are many broken links in the main README. Especially in the section [Where do I find new features and legacy repos?](https://github.com/robotology/whole-body-controllers#where-do-i-find-new-features-and-legacy-repos).

Here is a list of the ones that I have found:
- [exploit friction and walking controller](https://github.com/gabrielenava/whole-body-controllers)
- all links in [Legacy](https://github.com/robotology/whole-body-controllers#legacy)
- [idyntree-high-level-wrappers](https://github.com/gabrielenava/idyntree-high-level-wrappers)

aikolina, 2019-01-10:
**Related issue:** [Improve documentation](#55)

gabrielenava, 2019-01-10:
I think I only updated the links in the [noSimulinkGui-branch README](https://github.com/robotology/whole-body-controllers/blob/noSimulinkGui/README.md#where-do-i-find-new-features-and-legacy-repos), while the one in `master` are old. I think I will merge the PR, as the branch has been tested twice on the robot. I will tag the current `master`.

gabrielenava, 2019-01-10:
To avoid confusion I merged the `noSimulinkGui` branch in `master`. @aikolina now `master` branch contains the most up-to-date version of whole-body-controllers. (It is not that different, apart from the [wbc library](https://github.com/robotology/whole-body-controllers/issues/52) and the static GUI). I close this issue, as the links in the README should be fine now.

closed 2019-01-10 by gabrielenava