robotology/whole-body-controllers #61: Include motor reflected Inertia in the balancing controller

gabrielenava, 2019-02-14:
### References:
https://arxiv.org/abs/1803.04266

### Issue recap: how to add motor reflected inertia with coupled joints to the rigid body system dynamics

Consider the following equations of motion for a humanoid robot:

[image]

where (1) represents the floating base dynamics, (2) is the joint space dynamics, and (3) is the motors dynamics. The above equations represent a simplified model for the robot-motor dynamics under the following assumptions
* The angular motor kinetic energy is due to its own spinning only, and the center of masses of each motor is along the motor axis of rotation;
* Both stiffness and damping of the series elastic actuators can be considered linear versus its relative displacement absolute value and rate-of-change.
* All motors are rigidly connected to the transmission element.

Now, focus on Eq. (3):

-  Γ  is a diagonal matrix containing the gear box ratios;
-  T  is a block diagonal matrix, representing the the coupling between the input and the output rotations of the coupling mechanism;
- I_m is a diagonal matrix containing the motor inertias.

Then, as explained [here](http://wiki.icub.org/wiki/ICub_coupled_joints#From_kinematic_coupling_into_dynamic_coupling), the kinematic relation 

[image]

induces the following relationship between joint torques and the torques transmitted to the motor after gear box and coupling:

[image]

The above justifies the second term on the right hand side of Eq. (3).

The above kinematic relation, instead, induces the following relationship between motors and joint accelerations:

[image]
[image]

By substituting this last expression into (3) and multiplying by transpose((TΓ)^-1), one has:

[image]

Finally, by summing up (4) with (2) one has:

[image]

where:

[image]

[image]

Now, the system (5)-(6) can be solved using `u` as control input, and applying e.g. a classical  `momentum-based` approach. Then, the control input `u` can be related to the actual joint torques using (4) as follows:

[image]

where `K_ff`∈ [0,1].

### Documentation on how to create matrix T:

http://wiki.icub.org/wiki/ICub_coupled_joints

### Implemented in the controllers with [this commit]( https://github.com/robotology-playground/wholeBodyControllers/commit/b82317a0e9eeccdb7bdb3ba22e81449e2b4d43b6).

gabrielenava, 2019-02-14:
Pointers to the code:

https://github.com/robotology/whole-body-controllers/blob/master/torque-controllers/momentum-based-yoga/app/robots/iCubGazeboV2_5/initCoordinator.m#L86

https://github.com/robotology/whole-body-controllers/blob/master/library/matlab-wbc/%2Bwbc/computeMotorsInertia.m

https://github.com/robotology/whole-body-controllers/blob/master/library/matlab-wbc/%2Bwbc/addMotorsInertia.m

Unfortunately, these are only utility/configuration functions, as the motors reflected inertia have been integrated inside the Simulink model: 

https://github.com/robotology/whole-body-controllers/blob/master/torque-controllers/momentum-based-yoga/torqueBalancingYoga.mdl

closed 2019-02-14 by gabrielenava

jcarpent, 2019-02-14:
@gabrielenava Thank you for these nice explanations.
I'm surprised to not see in Eq. (3) the presence of Coriolis terms, like a Gyro would do. Is it a simplification for getting simpler equations?

gabrielenava, 2019-02-14:
If I'm not mistaken, eq. (3) should be correct provided that one of the modeling assumptions we did holds, in particular:

>   The angular motor kinetic energy is due to its own spinning only, and the center of masses of each motor is along the motor axis of rotation

which results in a constant and diagonal motors inertia matrix I_m. 

However, concerning the dynamic modelling with motors inertia, I just realized that there is another reference that can be very useful, in particular slides 10-17:
 
http://www.diag.uniroma1.it/~deluca/ADL_Dottorato2011_Part1_slides.pdf

jcarpent, 2019-02-14:
> If I'm not mistaken, eq. (3) should be correct provided that one of the modeling assumptions we did holds, in particular:
>
>
> which results in a constant and diagonal motors inertia matrix I_m.

Sounds reasonable for motors that are fixed with respect to a Galilean frame. 
But I would say this is not true for a motor embedded in a link which is rotating in the air. Do you agree this last observation?

gabrielenava, 2019-02-14:
I think I finally  got your point now :-) I reviewed the equations with the help of @traversaro and we come up with this explanation (I hope I'll be able to write it down in a clear and complete way):

Consider one single motor, and define the following reference frames:

- R is a frame attached to the rotor;
- I is a Galilean (inertial) frame;

then, if I got the problem, the equations representing the `single motor` rotational dynamics should  be defined as (I represented them in body (rotor frame R) coordinates to have constant inertia matrix):

[image]

where $\bar{\tau}$ is accounting for all the torques acting on the system, the angular velocity ${\omega}_{I,R}$ represents the angular velocity of the rotor w.r.t. the inertial reference frame I.

The second term on the left side of the equation is the one accounting for the fact that the motor is not fixed w.r.t. the frame I, and can rotate in the air. Therefore this is the missing term that disappeared in Eq. (3).

Recall now that the angular velocity of the rotor w.r.t. the inertial frame can be written as:

[image]

where ${\omega}{I,S}$ is the relative angular velocity between the stator and the inertial frame and ${\omega}_{S,R}$  is the relative angular velocity between the rotor and the stator.

When assuming that `the angular motor kinetic energy is due to its own spinning only`, we are implicitly simplifying the above equation as follows:

[image]

thus when calculating the angular velocity we are only considering the contribution of the rotor rotating w.r.t. the stator, neglecting that also the stator is rotating w.r.t. the inertial frame. This highlight that there is a nomenclature issue in my theory review, as I usually refer to **motor** but I guess the real meaning is **rotor only**. Looking at [De Luca slides](http://www.diag.uniroma1.it/~deluca/ADL_Dottorato2011_Part1_slides.pdf) it seems that this assumption is also done there, pag. 14.

Then, we substitute the expression we just found for the rotor's angular velocity, and we assume that the local frame R associated to the rotor has been chosen in such a way that the frame axis are along the principal axis of inertia and consequently the inertia matrix in body coordinates is diagonal. In this case and considering the above assumption, it holds that:

[image]

furthermore, the equations representing the `single motor` rotational dynamics can be reduced as:

[image]

where $i_{zz}$ is the third component along the diagonal of the inertia matrix. Replicating this equation for all the motors and stacking all the equations together leads to Eq. (3).

jcarpent, 2019-02-14:
Our details are totally clear. Thank you very much for taking this time.
Do we know how much these assumptions are correct? For a humanoid robot, for instance, I guess it will mostly depend on the maximal relative angular velocity.

gabrielenava, 2019-02-15:
>Do we know how much these assumptions are correct? For a humanoid robot, for instance, I guess it will mostly depend on the maximal relative angular velocity

No idea, but one interesting point may be that the assumption was originally done for a different use case, that was to represent and control the motors' dynamics in case of elastic joints. Maybe (but this is only my assumption) given that our scope is just to add motors reflected inertia to the joints' mass matrix, it is possible to relax/remove this assumption and just properly compute and include the missing terms in the equations. But a theory review is required in order to verify this hypothesis.

jcarpent, 2019-02-15:
Yes, sure. It would be interested indeed.
I will give a try and let you know. 

Thank you again for your time and very clear explanations.