robotology/whole-body-controllers #62: Support dynamics integration inside Simulink

gabrielenava, 2019-03-05:
In the [devel](https://github.com/robotology/whole-body-controllers/tree/devel/controllers/floating-base-balancing-torque-control) branch, I've modified all the Simulink controllers to isolate all the input/outputs from/to the Gazebo simulator:

[image]

This new configuration will be one day exploited for allowing the user to choose between performing the simulation with Gazebo or instead interface the controller's I/O using a `forward dynamics` block that will integrate the dynamics inside Simulink without the need of Gazebo.

gabrielenava, 2019-04-04:
@giulero let's discuss about this task today afternoon. Is is fine for you?

traversaro, 2019-04-04:
Related WB-Toolbox issue: https://github.com/robotology/wb-toolbox/issues/164 .

traversaro, 2019-04-04:
Ideally in the future it would be great to automatically generate a BlockFactory/FMI block for the "MOMENTUM BASE TORQUE CONTROL" model (see for example https://github.com/CATIA-Systems/FMIKit-Simulink), instead of just generating a complete executable of the complete simulation as we are doing now.

traversaro, 2019-09-18:
cc @nunoguedelha

gabrielenava, 2019-10-16:
For the moment, I will create a simple subsystem that:

- compute the `forward dynamics` for a floating base robot, assuming the joint torques are given by the controller;

- integrate the `forward dynamics`;

- can switch from one foot to two feet balancing, but only assuming contacts are rigid and ideal (no proper modeling of contacts like in Gazebo);

- only interface our over-used controller, i.e. the `yoga` controller.

The subsystem will be stored in the [library]() folder, so that it can be used as example for other controllers. A copy of the subsystem will be integrated in the `yoga` controller.

gabrielenava, 2020-02-19:
This has been now done due to unexpected necessities. There is now a [simulink-balancing-simulator](https://github.com/robotology/whole-body-controllers/commit/272941bcf18828c4306afd816faa9fa21e158add) in `devel` branch that implements forward dynamics integration in Simulink. It is a first version and some things can be improved, but it =can be used as baseline for developing more complex simulators in the future. Closing the issue.

closed 2020-02-19 by gabrielenava