robotology/whole-body-controllers #64: Document the naming convention for the variables saved in the Matlab workspace

gabrielenava, 2019-03-06:
In the new `devel` branch, I've changed the visualization part in the Simulink models and therefore also the variables that are saved in the Matlab workspace after stopping the model. This issue is to document the new naming convention.

gabrielenava, 2019-03-07:
| Variable name          | Fixed-base-joint-control | Floating-base-position-control | Floating-base-torque-control |
|------------------------|--------------------------|--------------------------------|------------------------------|
| jointPos_DATA          |             Y            |               Y                |               Y              |
| jointPos_Error_DATA    |             Y            |               Y                |               Y              |
| jointPos_Desired_DATA  |             Y            |               Y                |               Y              |
| tauMeasured_DATA       |             Y            |               X                |               Y              |
| tauError_DATA          |             Y            |               X                |               Y              |
| tauDesired_DATA        |             Y            |               X                |               Y              |
| gravTorque_DATA        |             Y            |               X                |               X              |
| jointReferences_DATA   |             Y            |               X                |               X              |
| baseAcc_des_DATA       |             X            |               Y                |               X              |
| jointAcc_des_DATA      |             X            |               Y                |               X              |
| jointVel_DATA          |             X            |               Y                |               Y              |
| jointVel_Error_DATA    |             X            |               Y                |               X              |
| jointVel_Desired_DATA  |             X            |               Y                |               X              |
| feetContactStatus_DATA |             X            |               X                |               Y              |
| KP_postural_diag_DATA  |             X            |               X                |               Y              |
| QPStatus_DATA          |             X            |               X                |               Y              |
| forces_l_sole_DATA     |             X            |               X                |               Y              |
| moments_l_sole_DATA    |             X            |               X                |               Y              |
| forces_r_sole_DATA     |             X            |               X                |               Y              |
| moments_r_sole_DATA    |             X            |               X                |               Y              |
| tau_star_afterSat_DATA |             X            |               X                |               Y              |
| baseVel_DATA           |             X            |               X                |               Y              |
| base_pose_DATA         |             X            |               X                |               Y              |
| pos_CoM_DATA           |             X            |               X                |               Y              |

gabrielenava, 2019-03-07:
Documentation on the naming convention added. Closing the issue as archived knowledge.

closed 2019-03-07 by gabrielenava

fjandrad, 2019-04-11:
@gabrielenava you can update with the jerk controller