robotology/whole-body-controllers #66: Add joystick to the state machine

gabrielenava, 2019-07-01:
At LAAS, the joystick is integrated in the Simulink controller so that it is possible to see from Simulink when the user presses a button. Here is the package: http://robotpkg.openrobots.org/robotpkg/hardware/joystick-genom3/

I don't know how much it is complicated to implement something similar for YARP, but I guess most of the work is already done as we already use the joystick for the walking. 

If we have access to the joystick from Simulink we may redesign the state machine allowing the robot to perform several demos, movements and so on at the user's command.

Off course it is low priority, but it may be nice to have it @DanielePucci. Maybe for the flying robot? And standup demo too!

cc @Yeshasvitvs @diegoferigo @Giulero

DanielePucci, 2019-07-01:
Nice catch @gabrielenava 

CC
@traversaro @diegoferigo

gabrielenava, 2020-02-19:
Moved to `backlog` as in the end it does not seem it is really useful for the `yoga` demo.

gabrielenava, 2021-04-19:
the activity is not going to be addressed in the near future, closing the issue for now

closed 2021-04-19 by gabrielenava