robotology/whole-body-controllers #70: Whole Body Controllers Release 2.5

gabrielenava, 2019-10-10:
The 2.5 release of whole-body-controllers will **NOT** change the current repo structure, but will mainly be focused on adding **new features**. It will be an ambitious release, so it will probably take some time to be achieved. Here is the list of changes:

- [] support data processing of data coming from Simulink
- [] support runtime plots
- [] extend the `wbc` library by collecting other utility functions from other repositories
- [] include the momentum jerk control
- [] include the friction exploitation control
- [] include joystick interface for the yoga
- [] include `matlab-multi-body-sim`
- [] automatize as much as possible the simulation procedure with Gazebo
- [] properly document or simplify the procedure for adding `wbc` library to the Matlab path
- [] include forward dynamics block in the library
- [] update documentation and wiki
- [] add `gotowholebodycontrollers`
- [] code cleanup and update

gabrielenava, 2020-03-12:
I've increased the priority of the release and of the associated issues, with the objective of doing a sprint and close the release.

gabrielenava, 2020-03-18:
## CHANGELOG V2.5 (to be copied in the release description)

### New features

- added a data processing pipeline #20 
- added a Simulink model which also integrates the system's dynamics #62

### Enhancements

- when a joint hits the limits or a spike is detected, the name of the joint is printed in a warning #81 
- it is possible now to skip recompilation of the model after a re-start when using the GUI #82 

### Documentation

- improved documentation for the release #55 
- added pictures of the Simulink models in the readmes #47 

### Installation

- now the `goToWholeBodyControllers.m` function is generated when calling `startup_WBC.m` #68 
- Improved  `startup_WBC.m` #75

gabrielenava, 2020-03-22:
Almost finished: when https://github.com/robotology/whole-body-controllers/pull/92 and https://github.com/robotology/robotology-superbuild/pull/367 will be merged, the `2.5` release will be done.

traversaro, 2020-03-23:
> 
> 
> Almost finished: when #92 and [robotology/robotology-superbuild#367](https://github.com/robotology/robotology-superbuild/pull/367) will be merged, the `2.5` release will be done.

Not a blocker for the 2.5 release (actually it is a modification that needs to be done **after the 2.5 release), but I think we should also modify the superbuild to install whole-body-controllers also when `ROBOTOLOGY_USES_MATLAB` is disabled.

gabrielenava, 2020-03-23:
## TODO after merging:

- Modify the superbuild to install whole-body-controllers also when ROBOTOLOGY_USES_MATLAB is disabled;

- Update the superbuild and WBC documentation accordingly;

- Consistency check for all the hyperlinks in the readmes;

- Add link to the new Simulink-only controller in the wiki.

gabrielenava, 2020-03-24:
>Not a blocker for the 2.5 release (actually it is a modification that needs to be done **after the 2.5 release), but I think we should also modify the superbuild to install whole-body-controllers also when ROBOTOLOGY_USES_MATLAB is disabled.

@traversaro I was addressing this point but I have some questions:

- Looking at the [CMakeLists](https://github.com/robotology/robotology-superbuild/blob/master/CMakeLists.txt#L160) of `robotology-superbuild`, there is also the option `ROBOTOLOGY_NOT_USE_SIMULINK`. Should I bring whole-body-controllers out of both the tags? do we still need the `ROBOTOLOGY_NOT_USE_SIMULINK` tag or it can be removed (it seems it is used only in that line);

- So if I install `whole-body-controllers` without Matlab on my pc, it should just clone the repo and install the home positions and the library. The option for autogeneration is OFF, but what happens if a user turn it on without Matlab? should we consider also this hypothesis?

- I think we should also document in the superbuild README that if WBC is installed without Matlab, its functionalities are reduced.

traversaro, 2020-03-24:
>  Looking at the CMakeLists of robotology-superbuild, there is also the option ROBOTOLOGY_NOT_USE_SIMULINK. Should I bring whole-body-controllers out of both the tags? do we still need the ROBOTOLOGY_NOT_USE_SIMULINK tag or it can be removed (it seems it is used only in that line);

Mhh, `ROBOTOLOGY_NOT_USE_SIMULINK` is useful for avoiding failures in BlockFactory if Matlab is compiled but Simulink is not, the fact that is used only there sounds like a regression. Probably the USES_MATLAB option of BlockFactory (see https://github.com/robotology/robotology-superbuild/blob/bdf3b7c5f4d7d2531e8c566058937558f23de649/cmake/BuildBlockFactory.cmake#L13) should depend on the value of `ROBOTOLOGY_NOT_USE_SIMULINK`. However, this is unrelated.

> So if I install whole-body-controllers without Matlab on my pc, it should just clone the repo and install the home positions and the library. 

Yes. The reason we did  not install it in the past even if it was harmless do to so in a system where `ROBOTOLOGY_USES_MATLAB` is disabled was to save clone time and space on disk, but given that it is now useful we can just download and  install it in any case.

> The option for autogeneration is OFF, but what happens if a user turn it on without Matlab? should we consider also this hypothesis?

No, we only care for the default options and for the options that are exposed in the superbuild CMake.

> I think we should also document in the superbuild README that if WBC is installed without Matlab, its functionalities are reduced.

You can probably mention that either in the ROBOTOLOGY_ENABLE_DYNAMICS or ROBOTOLOGY_USES_MATLAB docs.

gabrielenava, 2020-03-25:
[WBC readme](https://github.com/robotology/whole-body-controllers/blob/master/README.md#installation-and-usage) updated and [PR](https://github.com/robotology/robotology-superbuild/pull/370) opened. 

Once the PR will be finished, I'll close this issue and the 2.5 release will be officially concluded.

gabrielenava, 2020-03-26:
The PR has been merged. Closing, the release is officially concluded.

closed 2020-03-26 by gabrielenava