robotology/whole-body-controllers #75: Investigate if it is necessary to remove StartupWBC.m

gabrielenava, 2019-10-10:
I understood that adding the `wbc` library using the [startup file](https://github.com/robotology/whole-body-controllers/blob/master/config/startup_WBC.m) is not really straightforward, even if it is documented. In particular, it is not a straightforward procedure for who installed this repo through the `robotology-superbuild`, that is the majority of the users. Maybe we can find alternatives for adding the `wbc` folder to the matlab path.

gabrielenava, 2019-10-15:
I would like to have the possibility to independently configure WBC without the superbuild, while on the other hand we need to simplify the installation procedure in case the repo comes along with the `robotology-superbuild`. I think we can proceed as follows:

- add to the [startup_robotology_superbuild.m](https://github.com/robotology/robotology-superbuild/blob/master/cmake/template/startup_robotology_superbuild.m.in) file in the `robotology-superbuild` the path to [+wbc package](https://github.com/robotology/whole-body-controllers/tree/master/library/matlab-wbc/%2Bwbc) 
- document somewhere that when running the`startup_robotology_superbuild.m` file also the `+wbc package` is added to the Matlab path.

Menawhile, we also leave the [startup_WBC.m](https://github.com/robotology/whole-body-controllers/blob/master/config/startup_WBC.m) file in this repo, and in the README will be documented to use it only when the repo is **not installed** through the robotology-superbuild.

gabrielenava, 2020-03-18:
To comply with the workflow of the `robotology-superbuild`, I think it is better if the `+wbc` library is installed and then Matlab links to the installed files. Note that the very same pipeline is done in [mex-wholebodymodel]() and for the [iDyntree-high-level-wrappers](), while whole-body-controllers library still points the source code instead of the installed one. 

Documentation has also to be updated accordingly.

gabrielenava, 2020-03-18:
No, actually I think the current pipeline is better for handling also the installation of the `goToWholeBodyControllers.m` file. I'll open the PR to the robotology-superbuild.

gabrielenava, 2020-03-18:
PR open: https://github.com/robotology/robotology-superbuild/pull/367

traversaro, 2020-03-19:
> Menawhile, we also leave the startup_WBC.m file in this repo, and in the README will be documented to use it only when the repo is not installed through the robotology-superbuild.

Note that you can also think of installing the `startup_WBC.m` script, and  then call that from the robotology superbuild script. Just  a question: why we don't install the `+wbc` directory directly to the `<install_prefix>/mex` directory as we already do with YARP and Matlab so there is no need to modify the matlab path?

gabrielenava, 2020-03-19:
> Note that you can also think of installing the startup_WBC.m script, and then call that from the robotology superbuild script. Just a question: why we don't install the +wbc directory directly to [...]

I was going towards this direction (see the outdated comment https://github.com/robotology/whole-body-controllers/issues/75#issuecomment-600811140) because I don't like the fact that the library is not installed, however then I have no clue how to handle the `goToWholeBodyController.m` script.

That script should point the absolute path of the `source` folder, where the Simulink models are. So if we proceed as follows:

>Note that you can also think of installing the startup_WBC.m script, and then call that from the robotology superbuild script. 

I still need to somehow get the path to whole-body-controllers source folder inside the `startup_WBC.m` script, so that I can generate the `goToWholeBodyController.m` script. So I cannot get rid of the source path as commented [here](https://github.com/robotology/robotology-superbuild/pull/367#issuecomment-600885296).

gabrielenava, 2020-03-20:
I've decided:

1) install the library in the `mex` folder;
2) install `startup_WBC.m` in the mex folder for the `wbc not in the superbuild` workflow;
3) revert modifications of `startup_robotology_superbuild.m` (but the comments and the removal of mex-wholebodymodel);
4) add a separate script in the whole-body-controllers source folder to create the `goToWholeBodyControllers.m` script;
5) when launching   `startup_robotology_superbuild.m` or `startup_WBC.m`, print a message saying that there is the possibility to link the folder where `pathdef.m` is installed with the source folder of wbc through the `goToWholeBodyControllers.m` script, and for that you have to check inside the WBC source code;
6) document all these modifications in the READMEs.

gabrielenava, 2020-03-22:
With https://github.com/robotology/whole-body-controllers/pull/92/commits/ac4a43714f2ca5b6d6ea08618acc462c797dc6fc the `wbc` library is now installed in the `/install/mex` folder. All the documentation has been updated accordingly. The file [createGoToWBC.m](https://github.com/robotology/whole-body-controllers/blob/update-autogeneration/config/createGoToWBC.m) generates the `goToWholeBodyControllers.m` script. It is not installed, but just advertised inside the [startup_WBC](https://github.com/robotology/whole-body-controllers/blob/update-autogeneration/config/startup_WBC.m.in#L81) file.

closed 2020-03-23 by gabrielenava