robotology/whole-body-controllers #76: Simplify simulation workflow

gabrielenava, 2019-10-10:
Workflow  for testing the controller with Gazebo is log, it requires to manually start several devices and several minutes to perform one single test. It would be nice to automatize the workflow as much as possible.

traversaro, 2019-10-10:
Related issue: https://github.com/robotology/wb-toolbox/issues/164 .

gabrielenava, 2019-10-15:
A possible solution for now may be to design a `yarpmanager` application as @diegoferigo suggested. Then, the same application can also be used on the real iCub.

traversaro, 2019-12-02:
This plugin may permit to run the whole body controllers without launching an external wholebodydynamics: https://github.com/robotology/gazebo-yarp-plugins/pull/457 .

gabrielenava, 2020-03-16:
I've decided to postpone also this activity from the `2.5` release to the `3.0` release. 

The motivation is that to create the `yarpmanager` application requires a certain amount of work, as I need to create the files, then modify the `CMakeLists` accordingly, then testing the new feature at least in simulation for all the controllers and finally update the documentation.  Everything to be  done for a feature that:

- it has not been required by any WBC user, even if it would be nice to have it;
- it is not an upgrade or a bugfix.

Therefore there is no hurry for this feature. I will move it to the next WBC upgrade `3.0`, so the release time of `WBC 2.5` will decrease.

gabrielenava, 2021-04-19:
the activity is not going to be addressed in the near future, closing the issue for now

closed 2021-04-19 by gabrielenava