robotology/whole-body-controllers #79: Include whole-body current controller among the available controllers

gabrielenava, 2019-10-15:
The issue is a remainder of the activity @nunoguedelha is carrying on. Given his recent results with the current control framework, it would be nice to have a `whole-body` current controller among the available WBC controllers. 

Time and type of controller (fixed base of floating base?) still has to be defined. It would be nice if the current controller also implements the [friction exploitation framework](https://github.com/robotology-legacy/WBI-Toolbox-controllers/tree/whole-body-controllers-legacy/controllers/legacy/momentum-based-yoga-friction), currently still in `legacy`.

nunoguedelha, 2019-10-15:
Regarding the friction exploitation VS compensation feature, I have a proposal that I mentioned to @DanielePucci some time ago. I am documenting it [here](https://github.com/dic-iit/element_torque-control-via-current/issues/79).

gabrielenava, 2024-02-02:
in the end we did not migrate the friction exploitation controller to WBC. I don't think it make sense to schedule such activity at the moment. For now, closing as not planned.

closed 2024-02-02 by gabrielenava