robotology/whole-body-controllers #80: Migrate home-positions from codyco-modules

gabrielenava, 2019-10-29:
We need to migrate home-positions for the iCub robots (namely, these files: https://github.com/robotology/codyco-modules/tree/master/src/modules/torqueBalancing/app/robots) inside the WBC repository. These files require installation, therefore we need to:

- locate them in a proper folder in the WBC repo;
- add the `CMake` file for installing the files;
- document the necessity of installing these files in order to use the robot home-positions.

gabrielenava, 2020-02-19:
Done with this commit: https://github.com/robotology/whole-body-controllers/commit/36d905658c657681208dfb106dcf53703d9738d3

Installation tested with the robotology superbuild and seems working fine. It is in `devel` branch for now, but [PR](https://github.com/robotology/whole-body-controllers/pull/77) for merging in master is already opened.

NB: to keep the commit history readable, modifications of the `CMakeLists.txt` and of the `README` files are in a separate commit: https://github.com/robotology/whole-body-controllers/commit/144332b9f58461e2dfca96eed1f80de2d8cf7fa1

Closing the issue.

cc @traversaro

closed 2020-02-19 by gabrielenava

traversaro, 2020-02-19:
> Installation tested with the robotology superbuild and seems working fine. It is in devel branch for now, but PR for merging in master is already opened.

How did you tested that the files are correctly be found? As far as I can see, the robots are installed n `<prefix>/share/wbc`, and so you would need to install that dir to `YARP_DATA_DIRS`. To avoid having to deal with a new value to the env variables, I suggest to install the robots files in `share/yarp/robots`, so that we do not have do deal with a new env variable value for these files?

gabrielenava, 2020-02-19:
> How did you tested that the files are correctly be found? As far as I can see, the robots are installed n <prefix>/share/wbc, and so you would need to install that dir to YARP_DATA_DIRS. To avoid [...]

I run `yarp resource --find homePosBalancing.ini` after uninstalling the previous installed home positions and I was able to find the home positions in the correct path (<prefix>/share/wbc) so I concluded that this path was already included in the `YARP_DATA_DIRS` variable because of the `robotology-superbuild` settings. But it is ok for me to install in `yarp` folder instead.