robotology/whole-body-controllers #81: Print the joints which undergo spikes when the emergency stop on encoders ratio triggers

S-Dafarra, 2019-10-29:
The Yoga has an emergency stop when the encoder ratio is bigger than a threshold. This is pretty useful, even if, when it triggers, it would be helpful to understand which joint  triggered it and what was the entity of the spike

gabrielenava, 2020-03-17:
Here, the cleanest solution is to save to workspace the [vector that contains the check done on each single joint](https://github.com/robotology/whole-body-controllers/blob/master/library/matlab-wbc/%2Bwbc/checkSpikes.m#L34), then in the [function that runs when the model is stopped](https://github.com/robotology/whole-body-controllers/blob/master/controllers/floating-base-balancing-torque-control/stopTorqueControlBalancing.m) we add a warning message. in this way it is even possible to print the name of the joint undergoing spikes, which is clearer than just printing  the joint number. 

However, this requires to  add an extra output to the [check_spikes](https://github.com/robotology/whole-body-controllers/blob/master/library/matlab-wbc/%2Bwbc/checkSpikes.m#L1) function, and this means breaking the compatibility of the `WBC matlab functions library` w.r.t. older models.

I will do it, but this change has to be advertised somewhere.

gabrielenava, 2020-03-17:
Also the [check_range](https://github.com/robotology/whole-body-controllers/blob/master/library/matlab-wbc/%2Bwbc/checkRange.m) function can be updated in the same way.

S-Dafarra, 2020-03-17:
> However, this requires to add an extra output to the check_spikes function, and this means breaking the compatibility of the WBC matlab functions library w.r.t. older models.

I will do it, but this change has to be advertised somewhere.

gabrielenava, 2020-03-17:
>What about copying into a new one and deprecating the old one?

You convinced me, so let's do the following:

- I create two new functions `checkInputRange` and `checkInputSpikes`;

- I leave [checkRange](https://github.com/robotology/whole-body-controllers/blob/master/library/matlab-wbc/%2Bwbc/checkRange.m) and [checkSpikes](https://github.com/robotology/whole-body-controllers/blob/master/library/matlab-wbc/%2Bwbc/checkSpikes.m) in the library but I'll add a warning message saying that these functions are deprecated and they'll be removed in a future release;

- I will open an issue here as a reminder for removing the functions in the release `3.0`.

gabrielenava, 2020-03-18:
It was necessary to add a `unit delay` block because otherwise the `vector that contains the check done on each single joint` at time t = t_fail is not sent to workspace. But for the rest, I've implemented it.

After my testing session I'll add the commit and close the issue.

gabrielenava, 2020-03-18:
Tested. Here is the message when the joint hits the limits:

[image]

Here is the message when the joint has a spike:

[image]

Here is the part of the commit relevant for this issue (part of the modifications are hidden in Simulink models):

https://github.com/robotology/whole-body-controllers/commit/bdbf7fdb23600ba9bfa0cbb1b10b2b60daf8096b#diff-fc685c058894133c193c74a2f206badbL60

Same modification done for all the `stopFunctions`. Closing.

closed 2020-03-18 by gabrielenava

S-Dafarra, 2020-03-18:
Awesome!