robotology/whole-body-controllers #82: Avoid to recompile the model every time when using the static GUI

S-Dafarra, 2019-10-29:
Often, we need to run the YOGA from the static GUI multiple times in a row without changing anything. It is ok to compile the first time, but maybe after pressing STOP we could save some time by letting the user press START without compiling.

gabrielenava, 2020-03-12:
This logic was done because I was afraid about the possibility that the user starts the controller by mistake by clicking on the "start" button.

I think the best way is to add a check button. If it is checked, the user will not be forced to recompile the code.

So also the "safe" option (recompile every time before pressing start) is preserved.

gabrielenava, 2020-03-17:
I've implemented this request: now the GUI also has a checkbox as follows:

[image]

if you check it, the `start model` won't be disabled after stopping the simulation and therefore it will be possible to press it without the need of recompiling the model. I am currently performing a testing session before committing the changes, but it seems that the new logic is working fine and not conflicting with the previous workflow.

gabrielenava, 2020-03-18:
The new GUI has been tested with all models. It worked fine.

Here is the commit: https://github.com/robotology/whole-body-controllers/commit/bdbf7fdb23600ba9bfa0cbb1b10b2b60daf8096b

Here the parts of the commit relevant for this issue:

https://github.com/robotology/whole-body-controllers/commit/bdbf7fdb23600ba9bfa0cbb1b10b2b60daf8096b#diff-a084b794bc0759e7a6b77810e01874f2L34

https://github.com/robotology/whole-body-controllers/commit/bdbf7fdb23600ba9bfa0cbb1b10b2b60daf8096b#diff-df3f36a01ac9190881e2927bae2bbb7fL152

closed 2020-03-18 by gabrielenava