robotology/whole-body-controllers #83: Tune retargeting yoga

kouroshD, 2019-11-05:
According to https://github.com/dic-iit/component_wholebody-teleoperation/issues/194

> Tune RETARGETED YOGA

It is preferable to align the code to the whole-body-controller master branch after testing and tuning. We need to discuss it we @DanielePucci and @gabrielenava .

We will tune the smoothing parameters, so that the performance of retargeting increases.

kouroshD, 2019-11-05:
@gabrielenava  response:

> It is preferable to align the code to the whole-body-controller master branch after testing and tuning. We need to discuss it with @DanielePucci and @gabrielenava .

there is also a devel branch to which I commit new features. At the moment, the devel branch has:

updated documentation;
new offline automatic plotting of Simulink data;
other minor fixes.
There are also some feature request like robotology/whole-body-controllers#82 and robotology/whole-body-controllers#81 that I was thinking to implement in devel.

Maybe we can proceed as follows: we can have a look at the list of features I want to implement (robotology/whole-body-controllers#70 (comment)), then select the one you may need. I will implement them and merge a first PR from devel to master, and then you can align your code with the master branch.

Yeshasvitvs, 2019-11-06:
Yoga retargeting is currently available in my fork on a branch https://github.com/Yeshasvitvs/whole-body-controllers/tree/feature/yoga-retargeting

As discussed yesterday, it is better to have it in the upstream https://github.com/robotology/whole-body-controllers on the devel branch as @gabrielenava suggested in https://github.com/robotology/whole-body-controllers/issues/83#issuecomment-549724319

@kouroshD To proceed in this direction, the important commits containing the changes for yoga retargeting are from https://github.com/Yeshasvitvs/whole-body-controllers/commit/0fb6ae2653994c51292fb999e8113852368e8287 to https://github.com/Yeshasvitvs/whole-body-controllers/commit/81de363b54e0024cf5241f96292c3f4b9452c092. We can start with including these changes and test them first on greeny.

Once the above tests are done, we can include the changes from https://github.com/Yeshasvitvs/whole-body-controllers/commit/b3531a00d84ce1d382505d69c3dd3b0512a96fef to https://github.com/Yeshasvitvs/whole-body-controllers/commit/5bb02de307ce408743e17b97c1d12930eeb4e448 and update the CoM values for retargeting correctly.

gabrielenava, 2019-11-20:
Now the latest version of the yoga retargeting has been fine tuned and can work robustly both on the left and on the right foot. closing.

closed 2019-11-20 by gabrielenava

kouroshD, 2019-11-28:
The latest version of the code is located in this branch https://github.com/Yeshasvitvs/whole-body-controllers/tree/feature/kinematicSwitch