robotology/whole-body-controllers #84: Investigate how switch feet for yoga retargeting

kouroshD, 2019-11-05:
According to: https://github.com/dic-iit/component_wholebody-teleoperation/issues/194

> Insert the switch from one foot to two feet (either automatic, or triggered from kinematics or buttons)

Since, the ft shoes now are quite reliable we may go for that. Maybe also wireless shoes if possible. Something to e discussed with @DanielePucci . Other options can be using the joypads, or some threshold on the kinematic side.

Currently, the change from single support to double support is not tested and checked. We should do it after tuning.

DanielePucci, 2019-11-11:
Nice @kouroshD, wireless or wired shoes are a nice idea! Contact retargeting I guess is another open problem

gabrielenava, 2019-11-20:
Now the latest version of the yoga retargeting can switch from one foot to two feet balancing. Closing the issue.

closed 2019-11-20 by gabrielenava

kouroshD, 2019-11-22:
The implementation is in this branch https://github.com/Yeshasvitvs/whole-body-controllers/tree/feature/kinematicSwitch . we should do some additional tests, and add it to robotology branch.