robotology/whole-body-controllers #87: Migrate jerk control in whole-body-controllers

gabrielenava, 2020-03-12:
Following https://github.com/robotology/whole-body-controllers/issues/72, I will migrate the jerk control inside this repository.

I want to wait the [associated paper](https://arxiv.org/pdf/1907.11906.pdf) to receive its final evaluation before migrating the code, then I'll port the code in WBC.

traversaro, 2020-03-13:
> I want to wait the [associated paper](https://arxiv.org/pdf/1907.11906.pdf) to receive its final evaluation before migrating the code, then I'll port the code in WBC.

Just to understand, this is blocking WBC 2.5 release or not?

gabrielenava, 2020-03-13:
>Just to understand, this is blocking WBC 2.5 release or not?

@traversaro  No, it has been removed from the WBC 2.5 release. All the issues to be addressed for the 2.5 release have the `high priority` tag.

gabrielenava, 2021-04-21:
Jerk controller added on `devel` branch with this commit: https://github.com/robotology/whole-body-controllers/commit/5a1a92c6d3896980ca1b5f82d0e73ace3c06f33e

documentation updated. Closing the issue!

cc @nunoguedelha

closed 2021-04-21 by gabrielenava