robotology/whole-body-controllers #95: Improve the Simulink-balancing-simulator

gabrielenava, 2020-03-29:
The new `2.5` release of WBC now also officially supports an internal simulator for balancing: https://github.com/robotology/whole-body-controllers/blob/master/controllers/simulink-balancing-simulator/README.md.e

The simulator can be improved as follows:

- the control subsystem and the `dynamics integration` subsystem should be clearly separated. In this way the simulator may be easily adapted to other Simulink controllers;

- the blocks contained in the `dynamics integration` subsystem may be further separated in smaller subsystems and may become a small Simulink blocks library (to be added to https://github.com/robotology/whole-body-controllers/tree/master/library/simulink-library). I like this solution because I can replace these blocks if one day we decide to add to the WBToolbox library dedicated blocks for dynamics integration;

- at the moment we don't have a model of the environment: we assume the robot has 1 (or more than 1) link bolted to the ground. It would be nice to have a simple contact model to simulate the interaction between the foot and the environment. I don't need something really complex, so I ca search for solutions in the literature and try to implement them (cc @GiulioRomualdi);

- at the moment Gazebo is required to visualize the robot movements. I would like to have an internal visualizer that substitute Gazebo (cc @fjandrad).

fjandrad, 2020-04-01:
@gabrielenava  you can start experimenting with [the new matlab visualization]. Some documentation can be found [here](https://github.com/robotology/idyntree/blob/newMatlabVisualization/doc/matlab_visualization.md)

gabrielenava, 2021-04-19:
we could try to understand how to substitute this internal WBC simulator with [matlab-whole-body-simulator](https://github.com/dic-iit/matlab-whole-body-simulator) functionalities.

cc @nunoguedelha

nunoguedelha, 2021-05-06:
Hi @gabrielenava , as we discussed T2T some time ago, we could consider #121 as a first step in the strategy you defined here. Considering the bullet points you defined, I would say that #121 covers the bullets **1, 3** and **4**.

The point **2** would be the next step. I suggest we make this issue #95 an Epic and include #121 in it for starting. What do you think?

gabrielenava, 2021-05-07:
>The point 2 would be the next step. I suggest we make this issue #95 an Epic and include #121 in it for starting. What do you think?

Agreed, the title is already very generic and it can be easily include several issues.

gabrielenava, 2024-02-02:
only one issue was left in this epic, and it in completed. Closing the epic!

closed 2024-02-02 by gabrielenava