robotology/whole-body-controllers #96: Add KPI logic to evaluate performance and robustness of controllers

fjandrad, 2020-05-05:
It is possible to attempt to measure the performance and robustness of controllers by looking at certain relevant quantities, key performance indicators (KPI).

#### Performance
This is related to tasks we typically require the robot to do in a way or another. They can be compared to a reference value based on what the planner or controller required the robot to do to achieve a goal. 
There are:
- joint position, velocity and acceleration errors
- COM position  errors
- centroidal linear and angular momentum and momentum derivative errors
- end effectors cartesian position and orientation error ( feet and hand trajectories ) 

#### Robustness
In this case, we relate the KPI to the dynamic stability of the robot and how much we consider it maybe able to deal with uncertainties or disturbances.
These are:
- CoP inside each contact
- Friction constraints of each contact
- Contact wrench sum (CWS)
- DCM wrt to ZMP or CWS for 3D case ?

From these criteria, some are global ( CWS, DCM ) or local ( COP, Friction constraints ). Although local may be preferred to avoid certain scenarios were globally seems stable but locally is not or barely stable, it may be worth exploring their relevance as KPI in conjunction with local quantities.

fjandrad, 2020-05-12:
After some thought, I decided against putting the code for evaluation here. The reason is that although it may be almost directly linked to these controllers at the moment, with the implementation of a proper interface it may be more generic. 
For this reason, I created a [ different repo](https://github.com/dic-iit/element_controllers_evaluation).

cc @S-Dafarra @gabrielenava @DanielePucci @nunoguedelha

gabrielenava, 2020-05-12:
@fjandrad that's ok, but the repo you linked seems not accessible :-D.

Should we close the issue then?

closed 2020-05-12 by fjandrad