robotology/whole-body-controllers #109: Can we change the name of the repo?

gabrielenava, 2021-02-11:
I think that `whole-body-controllers` is a very generic title, also the controllers in https://github.com/robotology/walking-controllers are whole-body-controllers. 

This repo actually contains:

- MATLAB-Simulink -based, instantaneous whole-body controllers 
- a library of utility functions for MATLAB-Simulink -based, instantaneous whole-body controllers 
- some utility Simulink models/script for working both in simulation and on the real robot with MATLAB-Simulink based controllers.

Fun fact:  in `walking-controllers` there is a folder called `WholeBodyControllers`: https://github.com/robotology/walking-controllers/tree/master/src/WholeBodyControllers

will the change of name create big issues? 

cc @DanielePucci @traversaro @diegoferigo @S-Dafarra @Giulero @GiulioRomualdi @carlottasartore (I will add other people if necessary)

traversaro, 2021-02-11:
> will the change of name create big issues?

No, we just need change the name of a few files in the robotology-superbuild, i.e.:
* https://github.com/robotology/robotology-superbuild/blob/master/releases/latest.releases.yaml#L106
* https://github.com/robotology/robotology-superbuild/blob/master/cmake/Buildwhole-body-controllers.cmake
* https://github.com/robotology/robotology-superbuild/blob/3dc34724f6f4b41d74ede4c7d1b30021efe2fe20/cmake/RobotologySuperbuildLogic.cmake#L39
* An more in general any reference, see https://github.com/robotology/robotology-superbuild/search?p=2&q=whole-body-controllers 

So, if we want to, no problem. However, I personal prefer to avoid endless [bikeshadding](https://en.wikipedia.org/wiki/Law_of_triviality) discussion on naming, so as mantainer if you have a name that you prefer feel free to propose it, and if nobody objects we will change it.

traversaro, 2021-02-11:
As this will change the projects tags files, probably  @CarlottaSartore  needs to be informed in time.

CarlottaSartore, 2021-02-12:
Yes,  as long as we are informed in time for changing the tags  no problem from our side.

traversaro, 2021-02-12:
Note that the only reason why the name in the project tags will need to be changed is just for how the superbuild works, the old repo name would still work fine when doing git clone, as GitHub handles redirects quite well.

DanielePucci, 2021-02-12:
Go for me

gabrielenava, 2021-02-16:
After a f2f discussion with @CarlottaSartore we decided to postpone the update of the repo name in March. Also I realized that all the documentation which refers to whole-body-controllers needs to be updated and this is not so short.

traversaro, 2021-04-13:
> 
> 
> After a f2f discussion with @CarlottaSartore we decided to postpone the update of the repo name in March. Also I realized that all the documentation which refers to whole-body-controllers needs to be updated and this is not so short.

Is the renaming still planned?

gabrielenava, 2021-04-16:
>Is the renaming still planned?

I think it would be nice to align the name of the repo with the one of [matlab-whole-body-simulator](https://github.com/dic-iit/matlab-whole-body-simulator), and therefore change it to `matlab-whole-body-controllers`. I think the two repos may have things in common. Maybe we can take a final decision in today's meeting @traversaro.

gabrielenava, 2021-04-16:
In the end after listening to the feedback from @nunoguedelha I decided for `simulink-whole-body-controllers` because all controllers are Simulink models.

I am going to change just the name, I am not going to change the name of the library package (+wbc) otherwise all MATLAB/Simulink code that calls functions with the namespace `wbc` will be broken. So, the modification should only affect the documentation of some internal repos.

I think modifications are required in both the CMake and the documentation of the `robotology-superbuild`, please @traversaro could you help me in understanding what needs to be modified there?

Other repositories whose documentation will be affected:

- `whole-body-controllers`: @gabrielenava will handle it
-  iRonCub internal repos: @gabrielenava will handle it
-  other projects: to be discussed with @CarlottaSartore 
-  other elements/repos in the DIC team? (@lrapetti, @Yeshasvitvs, @S-Dafarra)
 - jerk control: it is going to be ported in whole-body-controllers.

nunoguedelha, 2021-04-21:
@gabrielenava , I've added this to the epic #113 as it is one of the actions listed there. Let me know if for any reason this is not ok for you.

gabrielenava, 2021-04-21:
>@gabrielenava , I've added this to the epic #113 as it is one of the actions listed there. Let me know if for any reason this is not ok for you.

fine to me @nunoguedelha!

traversaro, 2021-07-06:
@gabrielenava can we move this in Ice-Box if you are not going to work on this?

gabrielenava, 2021-07-06:
>@gabrielenava can we move this in Ice-Box if you are not going to work on this?

ok for me!

gabrielenava, 2021-08-06:
In the end, I decided to give up with this change of name, as time passed by,  and it seems to me a minor change that will result in a not small task - as we need to spend time to notify the name change, to update documentation, and so on. 

I close the issue, I will reopen it if and when there will be space for this modification.

closed 2021-08-06 by gabrielenava