robotology/whole-body-controllers #110: Integral quaternion

gsilano, 2021-03-23:
Hi everybody! I hope my question would be interesting and a source of inspiration even if this is not specifically meant to work in the iCub ecosystem. Currently, I'm facing the problem of integrating the first time derivative of a quaternion, the opposite of what has been implemented in [this MATLAB function](https://github.com/robotology/whole-body-controllers/blob/master/library/matlab-wbc/%2Bwbc/quaternionDerivative.m). 

Googleing the problem, I come up with numerous solutions. Above all, I found of interest what reported at this website: https://www.ashwinnarayan.com/post/how-to-integrate-quaternions.

I worked on this, but when I run a simple test script I realized there is something wrong somewhere since single-axis rotations reflect in the wrong way in the Euler angles. I attached the MATLAB scripts I used for the tests together with the implemented MATLAB functions. 

I hope that solving this problem would be useful for me and for the whole community.

[MATLAB_functions.zip](https://github.com/robotology/whole-body-controllers/files/6192039/MATLAB_functions.zip)

traversaro, 2021-03-24:
Hi @gsilano, thanks for sharing your solution to this problem.

> Hi everybody! I hope my question would be interesting and a source of inspiration even if this is not specifically meant to work in the iCub ecosystem.

Just to understand, from your issue it seems that you already solved your question, is that correct? In this case, if you just want to share something that you find cool with the iCub community, probably it may be more indicate (in the sense that it would reach more people) to post this in the Robotology Community Discussions (see https://github.com/robotology/community/discussions). cc @pattacini

pattacini, 2021-03-24:
Yep, this particular category works in this case: https://github.com/robotology/community/discussions/categories/show-and-tell.

gsilano, 2021-03-24:
Hi @traversaro and @pattacini! I will move the conversation there. Thanks!

closed 2021-03-24 by gsilano