robotology/whole-body-controllers #113: Create a library of controller blocks out of the multiple models in this repo and integrate them in those models

nunoguedelha, 2021-04-21:
This repository implements a multitude of controllers that can be connected either to Gazebo, either to the real robot iCub. In order to facilitate the reuse of such controllers on other humanoid robots or their integration in other simulation frameworks that implement specific robot dynamics simulators like [`matlab-whole-body-simulator`](https://github.com/dic-iit/matlab-whole-body-simulator), we should create a library of controller blocks out of the models implemented in this repository:
- This would avoid copy pasting the controllers implemented here every time we wish to integrate them in another framework.
- Such library would be available in the Simulink Library Browser interface.
- The whole-body controller blocks provided by the library could then be placed in a Simulink model like any other WBT block.
- Such controller blocks would be connected to an external "Configuration" block the same way as the WBT blocks.

Once the library created, we shall integrate its blocks in the controller models implemented in this repository.

A first example of integration in an external framework, which motivated this task, can be found in https://github.com/dic-iit/matlab-whole-body-simulator/issues/32. This is also in line with what @gabrielenava wanted to do in https://github.com/robotology/whole-body-controllers/issues/95. The simulator in https://github.com/dic-iit/matlab-whole-body-simulator matches the requirements listed by @gabrielenava in that issue.

nunoguedelha, 2021-04-21:
We held a kick-off meeting for this activity.

## Discuss about creating a library of controller blocks out of the multiples models in whole-body-controllers

### Meeting Minutes 16/04/2021

**Attendees:** Nuno Guedelha, Gabriele Nava

After discussing with Gabriele, we agreed on the following terms:
 
#### Actions for Gabriele:
 
1. Rename the repository whole-body-controllers.
2. Integrate the new controller (minimum jerk controller) in the framework.
3. Update the documentation
 
#### Actions for Nuno:
 
1. Define the granularity used in grouping the simulink blocks into library sub-system blocks. These sub-system blocks would populate the controllers library as well as the Kinematics and the Dynamics libraries.
2. Review it with Gabriele and Silvio.
3. Implement the libraries and replace the respective blocks in the main models using those controllers (`controllers` folders).
4. Each of the three libraries (controllers, Dynamics, Kinematcis) would be defined by a single *.slx file in the folder `library/simulink-library`.
5. The controller models in `controllers` folder, e.g. `whole-body-controllers/controllers/floating-base-balancing-torque-control`, are model examples integrating a controller block from the controllers library, eventually multiple blocks from the Kinematics or Dynamics libraries, with interface blocks for communicating either with the robot simulated on Gazebo, either with the robot simulated in `matlab-whole-body-simulator`, either with the real robot.
6. Add a mask to each controller for clearly defining its interface (`Config` structure etc).
 
Nuno shall open, in an epic, an issue for each step. (#113)

nunoguedelha, 2021-04-21:
@gabrielenava , could you assign this to me, as we agreed, such that I convert it to an epic and add the sub-issues? Thx

## IMPORTANT
The first step...
> Create the first controller block of the library: the YOGA++ controller. (High priority, dependency of [matlab-whole-body-simulator#32](https://github.com/dic-iit/matlab-whole-body-simulator/issues/32).

is top priority since we want to use the YOGA++ as a showcase for the `matlab-whole-body-simulator` simulator (https://github.com/dic-iit/element_software-engineering/issues/24) which deadline is the 30th April.

It shall be the first issue opened under this Epic.

CC @traversaro , @DanielePucci

nunoguedelha, 2021-04-21:
@gabrielenava could you please also assign #114 to me, and authorize me to add this repo to the board of https://github.com/dic-iit/matlab-whole-body-simulator.

traversaro, 2021-04-21:
> 
> 
> @gabrielenava could you please also assign #114 to me, and authorize me to add this repo to the board of https://github.com/dic-iit/matlab-whole-body-simulator.

Done!

gabrielenava, 2024-02-02:
hello @traversaro 

as nuno left the team some time ago and I am not aware of any plan for carrying on this activity, can I close this epic and all the associated issues for now?

traversaro, 2024-02-02:
Sure!

closed 2024-02-02 by gabrielenava