robotology/whole-body-controllers #127: Fix the YOGA on right foot in the model YOGA-controller+matlab-simulator

nunoguedelha, 2021-05-10:
The YOGA starting on the right foot is not working for now in the model YOGA-controller+matlab-simulator most probably due to the current tuning in states 8 and 9 of the controller state machine. To be analyzed.

nunoguedelha, 2021-05-10:
@traversaro , I had forgotten this one, I'll put in the last position of the Backlog. I think showing more clearly the switching in the demo is a good significant addition, although not really a priority. Let me know in case you have a different opinion.

traversaro, 2021-05-10:
Ok for me, but first let's try to wrap up a first release with the YOGA++ demo working, as we discussed last week.

nunoguedelha, 2021-05-10:
Sure, I just opened it for not getting it out of sight.