robotology/whole-body-controllers #136: Should iCub V2.5 feet size be equal on left/right foot?

lrapetti, 2021-12-02:
Assuming that:
- the position of the soles frame inside the foot is mirrored in left and right foot 
- the soles reference frames have the same orientation

Then I would expect the dimension of the foot defined on the sole frame:
https://github.com/robotology/whole-body-controllers/blob/44ccfa2567938d527406837cfd848775096594bc/controllers/floating-base-balancing-torque-control/app/robots/iCubGazeboV2_5/configStateMachine.m#L155-L156
would be different on right and left foot.

I would expect something like:
```
left_foot                    = [-0.07  0.12 ;    % xMin, xMax
                                -0.045 0.05 ];   % yMin, yMax  
right_foot                   = [-0.07  0.12 ;    % xMin, xMax
                                -0.05 0.045 ];   % yMin, yMax  
```
where y dimensions are flipped on one foot.

@gabrielenava are my assumptions wrong? Am I missing something?

gabrielenava, 2021-12-03:
I think you are right @lrapetti, I suspect the feet size was never corrected as the asymmetry in the y direction seems small and the yoga motion in particular was always conservative enough to not be affected by this error. Could you open a PR to fix the feet size?

lrapetti, 2021-12-03:
> Could you open a PR to fix the feet size?

Sure, I can handle it. Since the same variable `feet_size` is used both on left and right foot, we can keep the change minimum simply by using a symmetric value like:
```
 feet_size                    = [-0.07   0.12 ;    % xMin, xMax 
                                 -0.045 0.045 ];   % yMin, yMax 
```

what do you think?

gabrielenava, 2021-12-06:
>Sure, I can handle it. Since the same variable feet_size is used both on left and right foot, we can keep the change minimum simply by using a symmetric value like:
>
>feet_size                    = [-0.07   0.12 ;    % xMin, xMax 
                                 -0.045 0.045 ];   % yMin, yMax 
>what do you think?

fine to me :-)

lrapetti, 2021-12-07:
PR opened https://github.com/robotology/whole-body-controllers/pull/137

gabrielenava, 2021-12-09:
PR merged and closed, thanks @lrapetti! closing the issue.

closed 2021-12-09 by gabrielenava