robotology/whole-body-controllers #138: master and devel branches are diverging?

traversaro, 2021-12-09:
Hi @gabrielenava, 
  I was investigating https://github.com/robotology/robotology-superbuild/issues/948, and I noticed that currently the master branch and the devel branch of this repo are diverging. In particular, this is the situation (taken from https://github.com/robotology/whole-body-controllers/network ):
[image]

Beside this diverge, it seems that `v2.5.4` was released on devel, including some commits that are not included in master. Do you think we can do something to remove this divergence (if necessary, eliminating devel altogether)?

gabrielenava, 2021-12-09:
I think the problem is due to some recent PR that I merged on `master` directly, and not passing through `devel` first. I will investigate the differences between `devel` and `master`, but I agree in removing `devel` (if and when possible) and keep using `master` only.

gabrielenava, 2021-12-09:
I have merged `devel` into `master`, now I think that `devel` is not diverging anymore but it is just behind `master` of several commits:

[image]

if no-one uses the `devel` branch I think we can delete it.  Can you thumb up for approval @nunoguedelha @S-Dafarra @traversaro @lrapetti @CarlottaSartore (hope I have included all people who may be interested)?

gabrielenava, 2021-12-09:
created a new tag in case it is needed for the robotology: https://github.com/robotology/whole-body-controllers/releases/tag/v2.5.5

traversaro, 2021-12-09:
> 

By the way, master is already the branch used for Unstable branches in superbuild for the whole-body-controllers: https://github.com/robotology/robotology-superbuild/blob/master/cmake/ProjectsTagsUnstable.cmake#L31 .

gabrielenava, 2021-12-13:
>if no-one uses the devel branch I think we can delete it. Can you thumb up for approval @nunoguedelha @S-Dafarra @traversaro @lrapetti @CarlottaSartore (hope I have included all people who may be interested)?

one last call, then today afternoon I will delete `devel`. Let me know in case for some reasons you don't want me to delete the `devel` branch.

gabrielenava, 2021-12-13:
`devel` branch deleted. Closing the issue.

closed 2021-12-13 by gabrielenava