robotology/whole-body-controllers #144: Error running floatingBaseBalancingTorqueControlWithSimulator.torqueControlBalancingWithSimu

traversaro, 2022-03-20:
I tried to run the `floatingBaseBalancingTorqueControlWithSimulator` by just running `floatingBaseBalancingTorqueControlWithSimulator.torqueControlBalancingWithSimu` as documented in https://github.com/ami-iit/matlab-whole-body-simulator/tree/ac47383e3c04cd0cfd0ab11e2479b7991be9f491#eyes-checking-the-installation, and the simulation is failing with error:
~~~
Unable to resolve the name mwbs.Utils.Rp2H. 
~~~

This are the version of the dependency that are installed: 
~~~
(mwbs) C:\Users\STraversaro>mamba list
# packages in environment at C:\Users\STraversaro\AppData\Local\mambaforge\envs\mwbs:
#
# Name                    Version                   Build  Channel
ace                       7.0.6                h0e60522_0    conda-forge
assimp                    5.2.3                hc2aa0de_1    conda-forge
asttokens                 2.0.5              pyhd8ed1ab_0    conda-forge
backcall                  0.2.0              pyh9f0ad1d_0    conda-forge
backports                 1.0                        py_2    conda-forge
backports.functools_lru_cache 1.6.4              pyhd8ed1ab_0    conda-forge
blockfactory              0.8.3               h0e60522_44    robotology
boost-cpp                 1.74.0               h9f4b32c_7    conda-forge
bzip2                     1.0.8                h8ffe710_4    conda-forge
ca-certificates           2021.10.8            h5b45459_0    conda-forge
colorama                  0.4.4              pyh9f0ad1d_0    conda-forge
decorator                 5.1.1              pyhd8ed1ab_0    conda-forge
eigen                     3.4.0                h2d74725_0    conda-forge
executing                 0.8.3              pyhd8ed1ab_0    conda-forge
freeglut                  3.2.2                h0e60522_1    conda-forge
freetype                  2.10.4               h546665d_1    conda-forge
gsl                       2.7                  hdfb1a43_0    conda-forge
icu                       68.2                 h0e60522_0    conda-forge
icub-firmware-shared      1.24.0              h0e60522_44    robotology
icub-main                 1.24.0              h4f909f9_44    robotology
icub-models               1.23.0           py39h24871e4_2    conda-forge
idyntree                  5.1.0            py39hf30b944_0    conda-forge
intel-openmp              2022.0.0          h57928b3_3663    conda-forge
ipopt                     3.14.5               hf6be2e5_0    conda-forge
ipython                   8.1.1            py39hcbf5309_0    conda-forge
irrlicht                  1.8.5                h739eaf8_2    conda-forge
jasper                    2.0.33               h77af90b_0    conda-forge
jbig                      2.1               h8d14728_2003    conda-forge
jedi                      0.18.1           py39hcbf5309_0    conda-forge
jpeg                      9e                   h8ffe710_0    conda-forge
lcms2                     2.12                 h2a16943_0    conda-forge
lerc                      3.0                  h0e60522_0    conda-forge
libblas                   3.9.0              13_win64_mkl    conda-forge
libcblas                  3.9.0              13_win64_mkl    conda-forge
libclang                  11.1.0          default_h5c34c98_1    conda-forge
libdeflate                1.10                 h8ffe710_0    conda-forge
libffi                    3.4.2                h8ffe710_5    conda-forge
libflang                  5.0.0           h6538335_20180525    conda-forge
libiconv                  1.16                 he774522_0    conda-forge
libjpeg-turbo             2.1.1                h8ffe710_0    conda-forge
liblapack                 3.9.0              13_win64_mkl    conda-forge
liblapacke                3.9.0              13_win64_mkl    conda-forge
libode                    0.16.2               h415ef7b_9    conda-forge
libopencv                 4.5.5            py39h606fa8f_5    conda-forge
libosqp                   0.6.2                h0e60522_3    conda-forge
libpng                    1.6.37               h1d00b33_2    conda-forge
libprotobuf               3.19.4               h7755175_0    conda-forge
libsodium                 1.0.18               h8d14728_1    conda-forge
libtiff                   4.3.0                hc4061b1_3    conda-forge
libwebp                   1.2.2                h57928b3_0    conda-forge
libwebp-base              1.2.2                h8ffe710_1    conda-forge
libxcb                    1.13              hcd874cb_1004    conda-forge
libxml2                   2.9.12               hf5bbc77_1    conda-forge
libzlib                   1.2.11            h8ffe710_1013    conda-forge
llvm-meta                 5.0.0                         0    conda-forge
lz4-c                     1.9.3                h8ffe710_1    conda-forge
m2w64-gcc-libgfortran     5.3.0                         6    conda-forge
m2w64-gcc-libs            5.3.0                         7    conda-forge
m2w64-gcc-libs-core       5.3.0                         7    conda-forge
[... 37 lines cut ...]
tbb                       2021.5.0             h2d74725_0    conda-forge
tinyxml                   2.6.2                h2d74725_2    conda-forge
tk                        8.6.12               h8ffe710_0    conda-forge
tornado                   6.1              py39hb82d6ee_2    conda-forge
traitlets                 5.1.1              pyhd8ed1ab_0    conda-forge
tzdata                    2022a                h191b570_0    conda-forge
u-msgpack-python          2.7.1              pyh9f0ad1d_0    conda-forge
ucrt                      10.0.20348.0         h57928b3_0    conda-forge
vc                        14.2                 hb210afc_6    conda-forge
vs2015_runtime            14.29.30037          h902a5da_6    conda-forge
wb-toolbox                5.5.0               h0e60522_44    robotology
wcwidth                   0.2.5              pyh9f0ad1d_2    conda-forge
wheel                     0.37.1             pyhd8ed1ab_0    conda-forge
whole-body-controllers    2.5.6               h0e60522_44    robotology
xorg-libxau               1.0.9                hcd874cb_0    conda-forge
xorg-libxdmcp             1.1.3                hcd874cb_0    conda-forge
xz                        5.2.5                h62dcd97_1    conda-forge
yaml                      0.2.5                h8ffe710_2    conda-forge
yarp                      3.6.0           py39h2bb6a00_44    robotology
yarp-matlab-bindings      3.6.1               h0e60522_44    robotology
ycm-cmake-modules         0.13.0               h0e60522_0    conda-forge
zeromq                    4.3.4                h0e60522_1    conda-forge
zlib                      1.2.11            h8ffe710_1013    conda-forge
zstd                      1.5.2                h6255e5f_0    conda-forge
~~~

traversaro, 2022-03-20:
I guess that `matlab-whole-body-simulator` should install a `+mwbs.+Utils` directory, but I could not find it anywhere. Any idea on this @gabrielenava @nunoguedelha @VenusPasandi ?

traversaro, 2022-03-20:
After the fix https://github.com/robotology/whole-body-controllers/pull/145, now the simulation is failing with error:
~~~
MATLAB System block 'torqueControlBalancingWithSimu/Robot Visualizer/RobotVisualizer' error occurred when invoking 'setupImpl' method of 'mwbs.Visualizers.robotVisualizer.RobotVisualizer'. The error was thrown from '
 'C:\Users\STraversaro\AppData\Local\mambaforge\envs\mwbs\Library\mex\+mwbs\+Visualizers\+robotVisualizer\RobotVisualizer.m' at line 53
 'C:\Users\STraversaro\AppData\Local\mambaforge\envs\mwbs\Library\mex\+mwbs\+Visualizers\+robotVisualizer\RobotVisualizer.m' at line 26'.
Caused by:
Unable to resolve the name iDynTreeWrappers.loadReducedModel. 
Component: Simulink | Category: Block error
~~~

However, this is an error as the `whole-body-controllers`  conda package does not depend on `idyntree-matlab-bindings`, so it is actually a robotology-superbuild issue.

traversaro, 2022-03-20:
However, running again the simulation fails with:
~~~
Build failed because the build file name(s) exceed the Windows limit of 260 characters. Build from a working directory with a shorter path, to allow build files to be created with shorter filenames. "C:\Users\STraversaro\AppData\Local\mambaforge\envs\mwbs\Library\mex\+wbc\examples\+floatingBaseBalancingTorqueControlWithSimulator\app\robots\iCubGazeboV2_5\slprj\_sfprj\torqueControlBalancingWithSimu\_self\sfun\src\c1_torqueControlBalancingWithSimu.cpp"
Component: MATLAB Function | Category: Build error
Build failed because the build file name(s) exceed the Windows limit of 260 characters. Build from a working directory with a shorter path, to allow build files to be created with shorter filenames. "C:\Users\STraversaro\AppData\Local\mambaforge\envs\mwbs\Library\mex\+wbc\examples\+floatingBaseBalancingTorqueControlWithSimulator\app\robots\iCubGazeboV2_5\slprj\_sfprj\torqueControlBalancingWithSimu\_self\sfun\src\c1_torqueControlBalancingWithSimu.cpp"
Component: Simulink | Category: Model error
~~~

Instead, this was due to my current working directory. Moving to `C:\` worked fine.

VenusPasandi, 2022-03-21:
> I guess that `matlab-whole-body-simulator` should install a `+mwbs.+Utils` directory, but I could not find it anywhere. Any idea on this @gabrielenava @nunoguedelha @VenusPasandi ?

Hi @traversaro, `Utiles` is a class of `matlab-whole-body-simulators` and it should be installed if you installed a version of `matlab-whole-body-simulators` which is newer than the commit https://github.com/ami-iit/matlab-whole-body-simulator/commit/6078b4f2c1a874e2b48158998e311a9edf870bd5

traversaro, 2022-03-21:
> > I guess that `matlab-whole-body-simulator` should install a `+mwbs.+Utils` directory, but I could not find it anywhere. Any idea on this @gabrielenava @nunoguedelha @VenusPasandi ?
> 
> Hi @traversaro, `Utiles` is a class of `matlab-whole-body-simulators` and it should be installed if you installed a version of `matlab-whole-body-simulators` which is newer than the commit [ami-iit/matlab-whole-body-simulator@6078b4f](https://github.com/ami-iit/matlab-whole-body-simulator/commit/6078b4f2c1a874e2b48158998e311a9edf870bd5)

I see. This means that we did a release of `whole-body-controllers` (`2.5.6`, done in https://github.com/robotology/whole-body-controllers/releases/tag/v2.5.6) without a compatible release of `matlab-whole-body-simulator`. Probably we can close this issue as soon as we do a release of `matlab-whole-body-simulator` .

traversaro, 2022-03-21:
I updated the issue to reflect the actual problem.

gabrielenava, 2022-03-23:
I think it would be nice to implement a github action that runs a test, at least for this Matlab-only controller, and give feedback on the controller status to avoid such problems. I don't know if it is possible to design a similar test also for Gazebo-related controllers. 

There was already the idea of implementing CI in the repo long time ago: https://github.com/robotology/whole-body-controllers/issues/88

traversaro, 2022-03-23:
@VenusPasandi did something similar for an internal repo, and I think we can do something similar here. Changing the simulation time in https://github.com/robotology/whole-body-controllers/pull/146 from infinite to a finite number was actually meant to go in that direction.

gabrielenava, 2024-02-02:
the associated [PR}(https://github.com/robotology/whole-body-controllers/pull/145) is closed with this comment:

https://github.com/robotology/whole-body-controllers/pull/145#issuecomment-1073827480

>Hi @traversaro, I think this fix is not required. Actually, the function Rp2H is moved from the class State to Utiles in the commit https://github.com/ami-iit/matlab-whole-body-simulator/commit/6078b4f2c1a874e2b48158998e311a9edf870bd5.

I assume the issue can be closed, @traversaro feel free to reopen it if necessary!

closed 2024-02-02 by gabrielenava