robotology/whole-body-controllers #149: Broken URL in documentation

traversaro, 2022-07-12:
The link http://citeseerx.ist.psu.edu/viewdoc/download?doi=10.1.1.62.8655&rep=rep1&type=pdf in https://github.com/robotology/whole-body-controllers/blob/6e9a67623642d190427bace17d0b439bd2a13e8d/library/matlab-wbc/%2Bwbc/rotationalPID_acceleration.m#L52 is broken. I wanted to send it to @alaurenzi after discussing it at lunch, but apparently it does not work anymore.

gabrielenava, 2022-07-18:
actually, I clicked on it and worked for me:

[image]

it took long time to load because it is a 300 pages thesis. may be because of that that seemed broken?

traversaro, 2022-07-18:
No, the URL was definitely broken some days ago (it was redirecting to the citeseerx main page). Probably it was a a citeseerx temporary problem, now it is working also for me.

closed 2022-07-18 by traversaro